jyberg/Enhanced-Nextion-Library #2: setValue 2 or 4 bytes? - mega2560

shvarevbear, 2019-11-19:
Thank you very much for quick response. I had to modify NexVariable in order to send int32_t
or uinit32_t to Nextion as follows. This worked for ITEAD library as well. Otherwise only two bytes were sent. 
Is it correct? As far as I know itoa on Arduino works only for 16-bits numbers

bool NexVariable::setValue(int32_t number)
{
    //buffer size of 10 gets Mega2560 to hang up
    char buf[20] = {0};
    String cmd;
    //change itoa to ltoa
    ltoa(number, buf, 10);

jyberg, 2019-11-19:
hi,
yes I need to go thru itoa function call, I use primary NodeMCU as my dev baoad and there int data type is 32bit but in arduino it is 16bit number, so looks like ltoa is correct function in general as long seems to be 32bit value and for unsigned 32bit integer ultoa function also buffer size 10 is too small for max / min values it needs to be 12 chars. I will do corrections in some point.

shvarevbear, 2019-11-19:
Sorry, I submitted a second issue which turns out to be just a serial
traffic problem. Without admin rights I cannot delete it. Could you remove
it, please?

jyberg, 2020-01-18:
Corrected on version: 0.11.1

closed 2020-01-18 by jyberg