jyberg/Enhanced-Nextion-Library #4: Arduino Mega and Nextion

lenardic, 2019-12-17:
Hello,

Thank you for your updates. I try to use your libraries but I cant get it working on my Mega. 

I have this code 

`#include "Nextion.h" 

NexButton b1_test1          = NexButton(0, 5, "b1_test1"); 
NexButton bt0_test           = NexButton(0, 8, "bt0_test"); 
NexNumber n_test            = NexNumber(0, 1, "n_test");
NexText t_test                    = NexText(0, 3, "t_test");

char buffer[100] = {0}; 

NexTouch *nex_listen_list[] = 
{
    
    &b1_test1,
    &bt0_test,
    &n_test,
    &t_test,
    
    NULL
};

void b1_test1PopCallback(void *ptr)
{
    uint32_t number2 = 0;  
    n_test.getValue(&number2);  
   
    Serial.print("SONDA_1 :");
    Serial.println(number2);
         
}

void setup() {
    
    nexInit();
    Serial.begin(9600);
    
    b1_test1.attachPop(b1_test1PopCallback);
    
}

void loop() {
  nexLoop(nex_listen_list);

}`

When I look at the serial monitor I only get 0 as value from the number box.

Thank you for your help.

jyberg, 2019-12-18:
hi,
first if you use defauls nexconfig seting nextion library uses default serial interface, you don't need to call Serial.begin but when you debug on same time it can be that debug connection uses same serial port and can cause conflict. You can also enable nextion library debug messages by defining DEBUG_SERIAL_ENABLEn in nectconfig file. Also when you call n_test.getValue(&number2); you should check return value so that you know has getValue call succed or not. Also Nextinit return true/false indicating success of connection.

jyberg, 2020-01-18:
I did changes to readMessages functionality it not use aymore Serial/Software serial implementation, as it was not stable enough at least in SoftwareSerial side

closed 2020-01-18 by jyberg