jyberg/Enhanced-Nextion-Library #6: Page number is meaningless on text objects.

usane1, 2020-01-21:
Hi Jyberg!

First of all, thanks for the work!
I started to use your library few days ago. I didn't use the original library so i don't know how that works. I use yours because it is the more updated. The itead haven't been updated since years, so this one seemed to be the choice. The CPU i connect Nextion is an ESO8266 NodeMCU. I have several text object on a Nextion page and i have almost the same page as this one, so i copied the page and deleted, modified the different objects, but because of the copy the remained objects have the same ID and same name than the soruce page. I sending temperature values in text format to the object on the first page, it displays correctly, however if i change page the values that exists on the second page updates as well. So it seems the NexText pageID is somehow don't work. I don't know if it is the same with the other objects like numbers or variables, i haven't tested them.

jyberg, 2020-01-21:
hi,
Library works correctly nextion interface not take page id paremer when setting page value.  When setting the value of object variable send command is <component name>.<variable>=<value>
in this case e.g. temperatureValue.txt="25.5"
to overcome this interface problem: 

Create page object like:
// page 0 home
NexPage pHome(0, 0, "home");

then when defining object in page give reference to page like:
NexText tHomeTime(2,3,"time", &pHome);

Now enhanced library creates global object name when sending the command: <page name>.<object name>.<variable>=<value>

e.g.
home.time.txt="17:59"

This global name/variable support is one of many enhancements in library ;)

Other options call in nextion ui sendme in every page initialization and in esp8266 code keep track current active page, or you send sendme command to nextion to request current page id.

I hope that this helped to go forward and resolve the problem

usane1, 2020-01-22:
True! I didn't think about what the command that is sent to nextion. There is no page or objectID parameter in it, only name and value as you wrote it. I rename the other page objects or i handle the page problem in the controller.
Thanks for the enlightment :)

usane1, 2020-01-22:
Created page object, now works fine. Thanks again. I close the issue.

closed 2020-01-22 by usane1