jyberg/Enhanced-Nextion-Library #9: getValue and getText doesn't work

Francesco-CH, 2020-03-16:
Could not figure out yet in the code. But getValue and getText doesn't work. 

Connectivity to Nextion ok the setValue and/or setText works as expected. But read-back the set-values with get...... no success.

jyberg, 2020-03-16:
I don't know this type problem exist. I have used these functions without problems with arduino and esp boads. pleace share more precise problem description/fault if exist

Francesco-CH, 2020-03-16:
I'll dig into it. Actually no error message, warning or whatsoever.
Everything works setValue, setText,   push, pop events.
Only get doesn't. That's really weird.

Using the example sketch posted here with no loop / pop / push.
Set a value (50, works i see it on the screen because of the long delay) - get the value (result is 0 because i doesn't work), add +1 (obviously to 0) and set the value again (resulting in value 1).

Baud rate is on 9600. Quite long delays between operations to avoid timing issues.
Testing with original Arduino Nano Every and Nextion Touch NX4832K035_011.

==== 
#include "SoftwareSerial.h"
#include "Nextion.h"

SoftwareSerial SSerial(2,3);

// Declare a number object [page id:0,component id:3, component name: "n0"]. 
NexNumber n0 = NexNumber(0, 3, "n0");

void setup(void)
{
  SSerial.begin(9600);
    
/* Set the baudrate which is for debug and communicate with Nextion screen. */
  nexInit();
  delay(1000);

  n0.setValue(50);
  delay(500);

  uint32_t number;
  n0.getValue(&number);
  delay(500);

  number += 1;
  n0.setValue(number);
}

void loop(void)
{
}

Francesco-CH, 2020-03-16:
Run the whole stuff if DebugPrint activated:
Here the outcome. Looks like a lot of timeout errors in recv:

======
bkcmd=3
recv command err value: 1A
recvRetCommandFinished err
page 0
recv command timeout
recvRetCommandFinished err
connect

recvRetString[0,]

n0.val=50
recv command timeout
recvRetCommandFinished err
get n0.val
recvRetNumber err
n0.val=1
recv command timeout
recvRetCommandFinished err
======

Francesco-CH, 2020-03-16:
recvCommand(const uint8_t command, uint32_t timeout)

ends with:
Error 1A .... Invalid Variable (according to Nextion instructions).

Francesco-CH, 2020-03-17:
Hi jyberg,

found the issue(s) and a temporary workaround:
General number-format issue, mentioned somewhere in a forum, which results in a compiler error:
in nexHardware.cpp:
calling bool recvRetString(String &str, uint32_t timeout)
{
    ......
    uint16_t tmout=timeout*1000;
    uint16_t mixs=micros();
    size_t end{mixs+tmout};
    //size_t end{micros()+timeout*1000};
   ......
the command -->  size_t end{micros()+timeout*1000} <-- has a (internal) conversion error from uint32_t to uint16_t ..... using the inserted two variables (tmout and mixs) the error does not occur anymore.

Same issue on calling: size_t readBytes(uint8_t* buffer, size_t size, size_t timeout)

The Issue of the not working getValue is a timing issue in: nexNumber.cpp
here the workaround code:
bool NexNumber::getValue(uint32_t *number)  //Timing Issue between getObjGlobalPageName and sendCommand
{
   String cmd = String("get ");
    getObjGlobalPageName(cmd);
    dbSerialPrint("                             ");  //WAIT to let getObjGlobalPageName complete
    dbSerialPrint("                             ");  //WAIT to let getObjGlobalPageName complete
    cmd += ".val";
    sendCommand(cmd.c_str());
    return recvRetNumber(number);
}

What I've experienced so far:
1. the #define NEX_DEBUG_SERIAL_ENABLE         MUST        be activated in nexConfig.h
2. two dbSerialPrint must be performed before sendCommand with quit a long string in them. Only one alone doesn't work.

Assuming a timing issue between getObj..... and sendCommand I've tried delays up to 5 seconds with no success. Only this weird dbSerialPrints worked to allow a "getValue" from the Nextion. 

I'm running out of ideas to resolve this.  

I've not played around with the other getValues in nexNumber.cpp and other nex____.cpp yet.

Francesco-CH, 2020-03-17:
Sorry forgot to mention ... for     getText     the workaround doesn't work.

Francesco-CH, 2020-03-18:
For what reason ever today the workaround stopped working.  I'm close to give up.

Francesco-CH, 2020-03-18:
Close to giving up - but found - finally.

in nexHardware.cpp

millis() instead of micros() * 1000 here:
bool recvRetString(String &str, uint32_t timeout)
{
    str = "";
    bool ret{false};
    bool str_start_flag{false};
    uint8_t cnt_0xff = 0;
    uint8_t c = 0;
    //size_t end{micros()+timeout*1000};    // Francesco correction millis instead micros
    size_t end{millis()+timeout};

    //while (end > micros() && ret == false)
    while (end > millis() && ret == false)
    {
        while (nexSerial.available())
        {
            c = nexSerial.read();
            dbSerialPrint(c);

            if (str_start_flag)
            {
                if (0xFF == c)
                {
                    cnt_0xff++;                    
                    if (cnt_0xff >= 3)
                    {
                        ret = true;
                        break;
                    }
                }
                else
                {
                    str += (char)c;
                }
            }
            else if (NEX_RET_STRING_HEAD == c)
            {
                str_start_flag = true;
            }
            yield();
        }
        delayMicroseconds(20);
    }
    dbSerialPrintln("");
    dbSerialPrint("recvRetString[");
    dbSerialPrint(str.length());
    dbSerialPrint(",");
    dbSerialPrint(str);
    dbSerialPrintln("]");

    return ret;

Same in readBytes().

size_t readBytes(uint8_t* buffer, size_t size, size_t timeout)  /// Francesco Change not micros but Willis
{
    //size_t end{micros()+timeout*1000};
    size_t end{millis()+timeout};

    size_t avail{(size_t)nexSerial.available()};
    //while(size>avail && end > micros())
    while(size>avail && end > millis())

    {
        delayMicroseconds(10);
        avail=nexSerial.available();
    }
    size_t read=min(size,avail);
    for(size_t i{read}; i;--i)
    {
        *buffer=nexSerial.read();
        ++buffer;
    }
    return read;

jyberg, 2020-03-19:
hi, good in nano size_t is two bytes max value 65535. so most probably with micros there happens overflow in timeout calculation when using micros. now with millis that not happen ao easily. I need to add overflow case in the code.

craigs711btm, 2020-04-05:
im having the same issue, just wont compile at gettext

jyberg, 2020-04-05:
hi,
I made commit to correct this issue. I was not able to repeat the problem so can you try to new version and tell if it works now. I also changed timeout function parameter datatype from uint32_t to size_t

craigs711btm, 2020-04-05:
Iv found it will compile if I try it a few times but not first time, I’m very new to Arduino so not exactly sure where its going wrong. Now iv got it to compile I still cant the nextion screen to communicate with the Arduino so back to the drawing board  to do some more research Regards craig  Sent from Mail for Windows 10 From: Jyrki BergSent: 05 April 2020 20:08To: jyberg/Enhanced-Nextion-LibraryCc: craigs711btm; CommentSubject: Re: [jyberg/Enhanced-Nextion-Library] getValue and getText doesn't work (#9) hi,I made commit to correct this issue. I was not able to repeat the problem so can you try to new version and tell if it works now. I also changed timeout function parameter datatype from uint32_t to size_t

closed 2020-04-13 by jyberg