jyberg/Enhanced-Nextion-Library #12: bug in NexRtc::read_rtc_time()

iw2lsi, 2020-07-08:
Hi... 

  there are some typos in  NexRtc::read_rtc_time(char *time,uint32_t len) and similar... the **not** operator is missing.

ie:

   if(recvRetNumber(&var)) 

instead of:

   if(!recvRetNumber(&var))

I'm not sure about the added recvRetCommandFinished(200) function calls... on my Nextion NX1060P101-011C-I it's required but maybe there is something wrong elsewhere.

     Best Regards

           Giampaolo

-----------

bool NexRtc::read_rtc_time(char *time,uint32_t len)
{
    char time_buf[22] = {"0000/00/00 00:00:00 0"};
    uint32_t year,mon,day,hour,min,sec,week;
    String cmd;
	
    cmd = "get rtc0";
    sendCommand(cmd.c_str());
    if(!recvRetNumber(&year))
    {
        return false;
    }

    recvRetCommandFinished(200);
    
    cmd = "get rtc1";
    sendCommand(cmd.c_str());
    if(!recvRetNumber(&mon))
    {
        return false;
    }    

    recvRetCommandFinished(200);
    
    cmd = "get rtc2";
    sendCommand(cmd.c_str());
    if(!recvRetNumber(&day))
    {
        return false;
    }   

    recvRetCommandFinished(200);
    
    cmd = "get rtc3";
    sendCommand(cmd.c_str());
    if(!recvRetNumber(&hour))
    {
        return false;
    } 

    recvRetCommandFinished(200);

    cmd = "get rtc4";
    sendCommand(cmd.c_str());
    if(!recvRetNumber(&min))
    {
        return false;
    } 

    recvRetCommandFinished(200);
    
    cmd = "get rtc5";
    sendCommand(cmd.c_str());
    if(!recvRetNumber(&sec))
    {
        return false;
    }    

    recvRetCommandFinished(200);
    
    cmd = "get rtc6";
    sendCommand(cmd.c_str());
    if(!recvRetNumber(&week))
    {
        return false;
    }    

    recvRetCommandFinished(200);
    
    time_buf[0] = year/1000 + '0';
    time_buf[1] = (year/100)%10 + '0';
    time_buf[2] = (year/10)%10 + '0';
    time_buf[3] = year%10 + '0';
    time_buf[5] = mon/10 + '0';
    time_buf[6] = mon%10 + '0';
    time_buf[8] = day/10 + '0';
    time_buf[9] = day%10 + '0';
    time_buf[11] = hour/10 + '0';
    time_buf[12] = hour%10 + '0';
    time_buf[14] = min/10 + '0';
    time_buf[15] = min%10 + '0';
    time_buf[17] = sec/10 + '0';
    time_buf[18] = sec%10 + '0';
    time_buf[20] = week + '0';
    time_buf[21] = '\0';
    
    if(len >= 22)
    {
        for(int i=0;i<22;i++)
        {
            time[i] = time_buf[i];
        }
    }
    else{
        for(uint32_t i=0;i<len;i++)
        {
            time[i] = time_buf[i];
        }
    }   
  return true;
}

jyberg, 2020-07-08:
Corrected in 0.11.8 version

closed 2020-07-08 by jyberg

iw2lsi, 2020-07-16:
I've tested the new revision... unfortunatly it does not work... at least with my nextion controller it seems that a recvRetCommandFinished(200); is required after every "get rtc**n**"; 

  Giampaolo

jyberg, 2020-07-16:
Hi,

The logic is same as on otiginal lib version and when reading nextion instuction set I cant find Anu extra informaation tuhat is missing Grim logic. I don't Haven enhanced version so I can't test...

//Jyke

iw2lsi, 2020-07-16:
Hello Jyke,

   I can't say why but I need to place calls to recvRetCommandFinished()
here and there to have a reliable communication... especially when events
and pollings are mixed.

   I'll try to investigate it ASAP using a logic analyzer on the UART...

 BTW: below is my exact version of panel and firmware

Connected! Com:COM20,baudrate:9600,Model:NX1060P101_011C(CTP),firmware
Ver:S116,Device serial number:9A17340144490435,CPUID:10501,Flash
Size:131072000(125MB)address:0

Best Regards,

Giampaolo

iw2lsi, 2020-07-16:
Hi,

  I've solved the problem... AFAIK it seems to me that as events are
asynchronous, there must be some kind of buffering in order to handle them
properly.

On the arduino forum I've found this post from RayLivingston:

Nextion v0.9.0 Library With Working GUI Events
<https://forum.arduino.cc/index.php?topic=620821.msg4206486#msg4206486>

where a modified version of NexHardware is provided, implementing a queue
for the events...

Below is a fork of ITEAD library with RayLivingston changes:

https://github.com/johnseghersmsft/ITEADLIB_Arduino_Nextion

I'm testing it right now and it seems to work... I can now mix polling and
events without issues.

    Best Regards

             Giampaolo

jyberg, 2020-07-17:
Hi,

Good that you find the issue, did you modified library code? Can you suggest merge/change to my enhanced version so that change is available for all users, for all boards...?

//Jyrki.

<https://forum.arduino.cc/index.php?topic=620821.msg4206486#msg4206486>

https://github.com/johnseghersmsft/ITEADLIB_Arduino_Nextion

iw2lsi, 2020-07-17:
Hi Jyrki,

  I had to switch to another project today... but if you can wait for a
week or two, I'll be back and try to merge the changes in NexHardware into
your code...

  As far as I know the changes are limited and it should not be a problem
at all

         Best regards

              Giampaolo

jyberg, 2020-07-17:
Hi,

No problem this is hobby for me... I just have used so much time to got this lib work in some level... And I want to help so that other users it is much easier... Nextion will not help...

//Jyrki