jyberg/Enhanced-Nextion-Library #20: Vis command

douile17, 2021-01-30:
Hello ,

first thanks for your amazing work , it's just perfect to control an Nextion easily. 

But now i have a little problem, i'm try to hide / show an animation if a sensor is HIGH , and my question is : 
is it possible with your Lib to work with the VIS command from nextion ? 

best regards
Andrew

jyberg, 2021-01-30:
Hi,

NextObject includes your needed functionality:
    /**
     * Hide or Show componen on current page
     *
     * @param visible - true Show component, false Hide component 
     * @return true if success, false for failure
     */
    bool setVisible(bool visible);

I don't know what component you are using and is there more functional class available for your component but NextObject is general base class with base component functionality so if there in no component specific class you can build you own and define NextObject as base class for it, or use directly NextObject class.

closed 2021-01-30 by jyberg

douile17, 2021-01-30:
Hi ,

thank you for this quick answer :) I would like to hide an image with when
a sensor is  LOW

i will check you solution ;)

thanks a lot !

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Garanti
sans virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>