jyberg/Enhanced-Nextion-Library #21: currentPageIdCallback usage

pilaf78, 2021-04-01:
Hi ,

First off all , thank you for this awesome library :)

Now im blocked on an implementation of the currentPageIdCallback 
I have no idea how to implement that...
Should i give a pointer of function ? 

Can you give me an exemple of implementation ?

Many thanks for you help

jyberg, 2021-04-01:
hi,
- first initialize the nextion instance:
SoftwareSerial nexSerial(D2, D1); // RX, TX
Nextion *nextion = Nextion::GetInstance(nexSerial);

-CurrentPageId is callback function that you need initialize:
	// functional callbacks
	nextion->currentPageIdCallback = currentPageCallback;

- And callback function definition e.g...
void currentPageCallback(uint8_t pageId)
{
	if(currentPageId==pageId)
	{
		return;
	}
	currentPageId=pageId;
}

closed 2021-04-01 by jyberg

pilaf78, 2021-04-01:
Good its ok
Many thanks :)