jyberg/Enhanced-Nextion-Library #25: Portability

EyitopeIO, 2021-09-08:
Hi, I'm making plans to port your library to another platform. I need to write a display driver for OpenWRT. Could you point me to the parts of your library that are platform-independent? For example, the src folder looks like something I can build upon. What else should I note?

Thank you for the work you put into this library.

jyberg, 2021-09-08:
hi,
From portability point of vew I see only issue that you need replace serial/software serial communication basiaclly implement needed wrapper for serial communication for nexSerial in nexHardware, or re-implememt nexhardware for other board. The issue can be that OpenWRT seems to reguire some type of wrapper/emulator for serial communication? I'm not familar woth OpenWRT boards connectivity/functionality... hope this helps... And if you sucseea so that OpenWrt interface support can be behind of compiler flag with out huge modification to the otjer functionality I hope that we can merge changes here so that library comes more usable for other boarda also...

closed 2021-09-16 by jyberg

EyitopeIO, 2021-09-16:
Hi,

This is now at github.com/EyitopeIO/Arduino-Onion. I'm yet to do the licensing and acknowledgment stuff, but I'm already set to use the ported code. 

A makefile exists in the repo. Because it's a Linux interface, few Arduino names were not compatible. For example, <yourSerial>.read( )/write( ) had to be renamed to o_read( ) and o_write( ) since Linux uses those names as standard functions.

I am yet to build my application on top of this library. but at least basic functionality works. I can accept queued events from Nextion, however, callback functions don't seem to be called. Callbacks are defined and registered in main.cpp if you wish to have a look.

I will revisit this again soon.

jyberg, 2021-09-19:
hi,

I checked it... Have you thing to implament chagnes so that Onion and other arduino/esp boards are all compatible using #ifdefs?
also why you have implemented min, max, utoa, itoa functions as they are available in base C++ / std library. also yield can be found from std or directy from linux