jyberg/Enhanced-Nextion-Library #27: get pageid and synchro

australopitheque, 2022-02-22:
Hello
is there a way to know the current pageid in the library and to check that the nextion has received/returned the data?
because I have a synchonization problem between my arduino/nextion
in my listening with `'NexTouch *nex_listen_list[] = {&p1,&p2,etc...}' `and the other programs in the `void loop(void)`, the arduino doesn't see the buttons or pages where it is rendered.
it is as if the nextion was mute
no way to check if the nextion has been listened to?
it probably needs a listening delay to receive a valid response?

jyberg, 2022-02-22:
hi, yes there are. From nextion documentation: sendme Sends number of currently loaded page over serial (0x66 Return Data)
– number of currently loaded page is stored in system variable dp
– used in a page’s initialize event will auto-send as page loads
usage: sendme

I have used two technique, by requesting active page by sending sendme and other I have set in Nextion editor every page activation page id sending to board.

australopitheque, 2022-02-22:
okay so by adding this code like this, I can get the actual page by just calling it?
`uint8_t getCurrentPage(){
  next->sendCommand("sendme");
  next->currentPageIdCallback ;
  uint8_t temp[5] = {0};
  if (sizeof(temp) != next->readBytes((char *)temp, sizeof(temp))) return 0;
  if (temp[0] == 0x66 && temp[2] == 0xFF && temp[3] == 0xFF && temp[4] == 0xFF){
    uint8_t pageid = temp[1];
    return  pageid;
  }
  return 0 ;
}`

jyberg, 2022-02-22:
use this libary callback fuction to listen page id
/**
 * Current Page ID callback function
 * The device returns this data after receiving “sendme” instruction)
 * 
 *  uint8_t pageId
 */
 void (*currentPageIdCallback)(uint8_t);

I have use it lite this in my code:
	   if(err) // some communication error or page id lost etc,.
	   {
			delay(100);
			nextion->sendCommand("sendme");
			nextion->nexLoop(nex_listen_list);
	   }

australopitheque, 2022-02-23:
`void (*currentPageIdCallback)(uint8_t);
I have used it like this in my code:
if(err) // some communication error or page id lost etc,.
{
delay(100);
nextion->sendCommand("sendme");
nextion->nexLoop(nex_listen_list);
}
{
// functional callbacks
nextion->currentPageIdCallback = currentPageCallback; // currentPageCallbak is function pointer that is called when pageid is received from Nextiom
nextion->nextionReadyCallback = NextionReadyCallback;

// object callbacks
}` and 

void (*currentPageIdCallback)(uint8_t);

if(err) // some communication error or page id lost etc,.
{
delay(100);
nextion->sendCommand("sendme");
nextion->nexLoop(nex_listen_list);
}
but at what time :
`void (*currentPageIdCallback)(uint8_t);`

is called?
Do you have an example?

jyberg, 2022-02-23:
I use this InitNextionCallbacks funtion to initialize all nextion callbacks clobal function call bacs that are not tied to any object and after that other object spicific callbacks like: "bApOnOff.attachPop(bApOnOffCallback);" When nextion send data by itself push/pop/sendme/ready... all that is handled wia callbacks. Nextion send data is read using nextLoop( function call. It is recommended that nextLoop is called requrally e.g every time in main fuction loop or if not that ofter then every x milliseconds. So in this case my example want's to call it imediatelly after sendme command is send. So nextLoop call callback when it receives data from Nextion

australopitheque, 2022-02-24:
ok thanks for the explanations

closed 2022-03-09 by jyberg