jyberg/Enhanced-Nextion-Library #28: attashpush is unreachable.

australopitheque, 2022-02-22:
Hello
when I want to use attashpush, I get an error while in attachpop it works.
`  paramkelvin.attachPush(kelvinpush,&paramkelvin);`

The error message tells me that attashpush is unreachable.

`fonction "NexButton::attachPush" (déclaré à la ligne 50 de "C:\USERS\JACQU\DOCUMENTS\PLATFORMIO\PROJECTS\WS2812\.PIO\LIBDEPS\MEGA2560 V3\ENHANCED-NEXTION-LIBRARY\INCLUDE\NexButton.h") est inaccessible`

jyberg, 2022-02-22:
I enabled it on version 1.4.1. I have no idea why it was disabled in original Nextion libary version but now it is enabled...

australopitheque, 2022-02-23:
hello 
the attachepush is now working.
I just have to try to figure out how to make it work

`Nextion *next = Nextion::GetInstance(Serial); // HW serial`

because it can't communicate with my nextion
in 
`Nextion *next = Nextion::GetInstance(mySerial); // software serial`
it works but only on non HW ports.

I can't find how to modify on `nexconfig.h `and assign the port for the nextion on RX/TX hardware

jyberg, 2022-02-23:
When you request nextion instace you define HardwareSerial or SoftwareSerial, I have mainly used SoftwareSerial as my espn has no free HardwareSerial port

SoftwareSerial nexSerial(D2, D1); // RX, TX
Nextion *nextion = Nextion::GetInstance(nexSerial);

if not available check that you have set #define  NEX_ENABLE_HW_SERIAL  as all different serial options are behind of #ifdefs

australopitheque, 2022-02-24:
I found it, it was as simple as that ^^
I had to declare it before.

```
#ifdef ESP8266
// esp8266 / NodeMCU software serial ports
SoftwareSerial mySerial(D2, D1); // RX, TX
#else
//SoftwareSerial mySerial(15,14); // RX, TX
//HardwareSerial mySerial(Serial3)  ;
#endif
Nextion *next = Nextion::GetInstance(Serial3); // HW serial
//Nextion *next = Nextion::GetInstance(mySerial); //   software serial
```

thank you very much for your help

closed 2022-03-09 by jyberg