jyberg/Enhanced-Nextion-Library #30: variable 'c' set but not used

australopitheque, 2022-05-05:
Hello
on platformIO with VScode I get an alert message at compile time in NexHardware.

`variable 'c' set but not used [-Wunused-but-set-variable]`

` [{
	"resource": "/Enhanced-Nextion-Library/src/NexHardware.cpp",
	"owner": "cpp",
	"severity": 4,
	"message": "variable 'c' set but not used [-Wunused-but-set-variable]",
	"startLineNumber": 729,
	"startColumn": 21,
	"endLineNumber": 729,
	"endColumn": 21
}]`

jyberg, 2022-05-05:
have you read the code? Variable c is used for debug message and readin data from buffer, in next lines. False Positive warning

closed 2022-05-05 by jyberg

australopitheque, 2022-05-05:
Hello
no I didn't look at it because this part in general I don't touch it, and as it doesn't block me, I just wanted to point out the warning when compiling...