jyberg/Enhanced-Nextion-Library #36: Update version in "library.json" and "library.properties"

maxus-git, 2024-04-30:
Hi Jyrki,

can you please update also the `library.json` and `library.properties` to version 1.5.0 (still on 1.4.3) so that the library can be updated also in Platform.IO?

Thank you!

Best regards,
Markus

jyberg, 2024-05-02:
Thanks, vesion not updated to 1.5.1 it should be available on Platform.IO after some time...

closed 2024-05-02 by jyberg

maxus-git, 2024-07-11:
Hi Jyrki,

PlatformIO still shows the last version with 1.4.3 --> The registry at "https://registry.platformio.org/libraries/jyberg/Enhanced-Nextion-Library" is not up to date with Github.

jyberg, 2024-07-16:
I did again re-publish now it looks like it is finally updated on platformio...

maxus-git, 2024-07-16:
Hi Jyrki,
thank you - looks good now.