jyberg/Enhanced-Nextion-Library #39: nextion lcd with esp32

fawadbutt, 2025-02-14:
Hi, i try to run esp32 with nextion lcd , with hardware serial 2 , my nextion code sucessfuly work with esp8266 but not working with esp32 , can you sugggest me the solution.? here is my code
`#include "Nextion.h"

// Nextion display initialization
Nextion* next = Nextion::GetInstance(Serial2);
NexPage page0(next, 0, "page0");
NexPage page1(next, 1, "page1");
NexPage page2(next, 2, "page2");
NexPage page3(next, 3, "page3");
NexPage page4(next, 4, "page4");
NexText gameName(next, 1, 1, "t0", &page1);
NexText gamePrice(next, 1, 2, "t1", &page1);
NexText cardBalance(next, 4, 1, "t0", &page4);

char machineName[50] = "Unknown"; // Default value to avoid null pointer issues
int machinePrice=0;
int balance = 30;
void setup() {
Serial.begin(9600);//
Serial.println("Initializing...");

if(!next->nexInit(19200)){Serial.println("nextion init fails");}
delay(5000);
Serial.println("--->Device Init Screen  Count-1");/// main Page
page0.show();
}

void loop(){
            page0.show();
            delay(2000);
            page1.show();
            gamePrice.setText(String(machinePrice).c_str());
            gameName.setText(machineName);
            delay(2000);
            page2.show();
            delay(2000);
            page3.show();
            delay(2000);
            page4.show();
            delay(2000);
            page4.show();
            cardBalance.setText(String(balance).c_str());
            delay(2000);
            page3.show();
            delay(2000);
            page2.show();
            delay(2000);
            page2.show();
    
}    `

jyberg, 2025-02-15:
hi, I have faced same issue. I think that in my board serial 2 ws used internally in board. 

So port availability depeds from used board...
this is quote good discussuion... https://forum.arduino.cc/t/esp32-uart-hardware-serial-ports-help/1011812/27

Also there is example of port test sowtware... I have used software serial library quite often it has worked fine...

fawadbutt, 2025-02-15:
I successfully run nextion with manul cammand. So there so issue in esp32 serial 2. I think issue in lib.  Serial2 16 and 17 working perfectly fine for me

closed 2025-02-15 by fawadbutt